bombcheck/ESP8266_ArtNet-LED-DMX-Node #1: nodeMCU fried. This is probably not a firmware issue but Ilooking for help.

potentstudios, 2018-09-13:
First off, thanks so much for this version! It solved the saving issues I was having before with Mtongnz's version. But like most things, you find one solution and another problem arises. I don't know if this is a firmware issue or not but I'm a bit puzzled so I figured I'd post here and see if anyone has any insight.

I was using the [usb charger](https://www.amazon.com/gp/product/B00P936188/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) to power 6 nodeMCUs for 6 separate 60 pixel long led strips and 3 of them got fried. I was using Resolume lighting software at 60fps (composition was 107x60px) and all 6 were working fine for about an hour then some started dropping off and now aren't detected by my PC and there's a bit of a smell coming from them. Before they died, when I used a specific clip, some started to drop out, then I'd switch to another clip and they would slowly go back to normal. [Here's a video of the issue.](https://www.youtube.com/watch?v=BU_3DBn8szg)

I also did a few stress tests to make sure the saving issues were really resolved and to simulate power issues that could arise in the field. I would unplug the usb power supply and re-plug in it. once fairly quickly. In hindsight this probably wasn't the best idea.

Is this power supply not recommended? Or maybe the nodeMCU can't handle the information at 60fps? I have a 470 ohm resistor on the data line. Maybe my unplugging them did something? After the quick unplugging and re-plugging back in the power, 2 of the strips stayed randomly spotty with blue and green for about 20-30 seconds which made me think I fried the led strips but then they went back to normal.

ahlynsky, 2018-09-14:
It’s your power supply.  Try running each with its own 2.4 amp supply, or all from a meanwell 60 amp supply and you should be good.  Try turning down the brightness and play only a single RGB color and I bet you will be fine.

potentstudios, 2018-09-14:
Thanks for the reply! So what caused them to fry if my supply wasn't powerful enough? These lights are for stage lighting so I need to be able to use max brightness and full color.

ahlynsky, 2018-09-14:
Yeah, probably each channel of that usb power supply thing could push 2.4 amps, but when you did them all white all at once the spike was too much.  You should check the peak draw with a multimeter.  What were the ratings on the brick feeding the usb hub?