bombcheck/ESP8266_ArtNet-LED-DMX-Node #3: hotspot password invalid

spinjim, 2019-05-26:
successfully flashed to a nodemcu. Shows up in my wifi list as "espArtLeDNode0203" but it says the password is invalid even when i use cut/paste (no quotes) "espArtLeDNode".

bombcheck, 2019-05-27:
Strange... Just checked within the code: The password is "espArtLeDNode" (without quotes).
Have you tried powercycling the nodemcu? Also, it may take some tries to connect to the wifi in this stage. Is your wifi adapter set toe dhcp?

spinjim, 2019-05-28:
Hi Nils,

Thank you so much for getting back to me so quickly.

I had no luck. I tried 2 nodemcu, I tried a Mac and a windows 7 machine.  Seems like it shows up in WiFi, then I try to connect by putting in the password. On both Mac and windows it tries to connect it even seems to connect then boom it just drops and I’m not connected.

I’ll give it another try hopefully next weekend. 

I think I purchased some software from you in the mid 90s. I think it was a guitar tuner for my Mac color classic. Or maybe it was zx80 stuff, it was long ago.

Thanks,

Jim