bombcheck/ESP8266_ArtNet-LED-DMX-Node #9: Can't Compile

sheddo01, 2020-10-04:
Hi,

Firstly, I'm quite new at this, so i am doing something dumb, Please feel free to point that out. Also I'll offer a blanket apology in advance for any dumb questions.

I am trying to compile the espArtLeDNode.ino sketch and I am getting an error:

ajax:103:55: error: ambiguous overload for 'operator~' (operand type is 'IPAddress')
       deviceSettings.broadcast = deviceSettings.ip | (~deviceSettings.subnet);

I have not altered anything in the original sketch

I have the libraries from your GIT installed.

I have updated pretty much all other libraries

Hardware side:
I am using a LOLIN V3 NodeMCU Clone, (I assume!)
I have my Board set to Generic ESP8266 Module
I have built a breadboard per the Original Schematics.

If you have any suggestions/recommendations or need further information please let me know.

Cheers

sheddo01, 2020-10-05:
Small Addendum!

I googeld the error message, and I saw a suggestion on another forum (Arduino forum) to change the ESP8266 in board manager to version 2.4.2.

Seems to work. I can compile and upload now.

Cheers

SpillzyAU, 2023-05-31:
Thanks Sheddo Guy!!! I owe you Lunch!!